earlephilhower/arduino-littlefs-upload #3: Show "Board details not available. Compile the sketch once." when click upload

Metaphorme, 2023-10-24:
Sorry for bothering you again...

I had set fqbn and I could compile and upload my firmware. But when I tried to update files into LittleFS, it shows "Board details not available. Compile the sketch once.".

I tried to know what it means and found something [here](https://github.com/earlephilhower/arduino-littlefs-upload/blob/main/src/extension.ts#L74) :

```typescript
if ((arduinoContext.boardDetails === undefined) ||  (arduinoContext.fqbn === undefined)){
    vscode.window.showErrorMessage("Board details not available. Compile the sketch once.");
    return;
}
```

I'm sure that I have set my fqbn, but I don't know how to set `arduinoContext.boardDetails`. Maybe this problem has to do with [this](https://github.com/dankeboy36/vscode-arduino-api/issues/6)?

fqbn had set:

[image]

boardDetails had set(maybe):

[image]

Tools -> Get board info:

[image]

Thanks for your help! 

Have a nice day! :)

earlephilhower, 2023-10-24:
Please see the readme about the bug in the IDE and workaround when this happens.

Metaphorme, 2023-10-24:
Solution: https://github.com/earlephilhower/arduino-littlefs-upload#glitches

Thanks! It never occurred to me that these two things could be so mysteriously related.

closed 2023-10-24 by Metaphorme

reivaxy, 2023-11-25:
@Metaphorme to me neither, thanks for creating this issue!
@earlephilhower It might help to add the error message "Board details not available. Compile the sketch once." in the glitches paragraph, what do you think?

earlephilhower, 2023-11-28:
@reivaxy I'm sorry, but I don't quite understand your suggestion.  Why not make a PR (for text docs like this you can do it right from the GH website) to make it clear?

Quantummechanic1, 2025-03-11:
This issue still exists in Arduino IDE 2.3.4 

[image]

@earlephilhower the suggestion by @reivaxy is to add the error message in the readme, so that a Google search will lead to that particular paragraph.

BTW, I opened the IDE (on Win10) by double clicking on the .ino file. The sketch compiled and uploaded successfully, but the Upload LittleFS failed with "Board details not available. Compile the sketch once.". 
A re-compile did not solve. A change of board and back and re-compile, as suggested, resulted in another error (sketch path and data path undefined).
For me, the problems were solved by opening another sketch (via File->Open recent), close the failing sketch, re-open and close the other sketch (i.e. there is finally only one IDE running).