earlephilhower/arduino-littlefs-upload #5: Board details not available. Compile the sketch once.

boomsya, 2023-11-23:
hi. Can you help me with error "Board details not available. Compile the sketch once." ?

**using Arduino 2.2.1 + nodemcu v3 (esp12-f)**
firmware compiled and uploaded. I tried to recompile, re upload but still have error "Board details not available. Compile the sketch once." :(

source code of plugin shows error in two cases:
if ((arduinoContext.**boardDetails === undefined**) **||** (arduinoContext.**fqbn === undefined**)) {
    vscode.window.showErrorMessage("Board details not available. Compile the sketch once.");
    return;
}
I dont know how I can verify which of this variables is undefined :(
I have some time to make tests. If you want we can try to find a solution.

[image]
[image]

boomsya, 2023-11-23:
after some experiments I have understand what lack of data arduinoContext.boardDetails and arduinoContext.fqbn
is a because you using other board library.
see what I am using:
[image]
after that I have replaced to:
[image]
and I have some other results in console:
```
{
	"sketchPath":"Z:\Development\Dryer\DIY-Filament-Dryer",
	"compileSummary":
	{
		"buildPath":"D:\temp\arduino\sketches\0FC4F4D26E59E77276FC5A3C9EC89556",
		"buildProperties":
		{
			"id":"nodemcuv2",
			"build.arch":"ESP8266",
			"build.board":"ESP8266_NODEMCU_ESP12E",
			"build.core":"esp8266",
			"build.core.path":"C:\Users\Niko\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266",
			"build.debug_level":"",
			"build.debug_optim":"-Os",
			"build.debug_port":"",
			"build.eeprom_start":"",
			"build.exception_flags":"-fno-exceptions",
			"build.extra_flags":"",
			"build.f_cpu":"80000000L",
			"build.flash_flags":"-DFLASHMODE_DIO",
			"build.flash_freq":"40",
			"build.flash_ld":"eagle.flash.4m2m.ld",
			"build.flash_mode":"dio",
			"build.flash_size":"4M",
			"build.float":"-u printf_float -u scanf_float",
			"build.fqbn":"esp8266:esp8266:nodemcuv2:led=2,baud=115200,xtal=80,eesz=4M2M,dbg=Disabled,lvl=None_,optim=Smallest,ip=lm2f,vt=flash,exception=disabled,stacksmash=disabled,wipe=none,ssl=all,mmu=3232,non32xfer=fast",
			"build.led":"-DLED_BUILTIN=2",
			"build.library_discovery_phase":"0",
			"build.lwip_flags":"-DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0",
			"build.lwip_include":"lwip2/include",
			"build.lwip_lib":"-llwip2-536-feat",
			"build.mcu":"esp8266",
			"build.mmuflags":"-DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000",
			"build.non32xferflags":"",
			"build.opt.flags":""@D:\temp\arduino\sketches\0FC4F4D26E59E77276FC5A3C9EC89556/core/build.opt"",
			"build.opt.fqfn":"D:\temp\arduino\sketches\0FC4F4D26E59E77276FC5A3C9EC89556/core/build.opt",
			"build.path":"D:\temp\arduino\sketches\0FC4F4D26E59E77276FC5A3C9EC89556",
			"build.project_name":"DIY-Filament-Dryer.ino",
			"build.rfcal_addr":"0x3FC000",
			"build.sdk":"NONOSDK22x_190703",
			"build.source.path":"Z:\Development\Dryer\DIY-Filament-Dryer",
			"build.spiffs_blocksize":"8192",
			"build.spiffs_end":"0x3FA000",
			"build.spiffs_pagesize":"256",
			"build.spiffs_start":"0x200000",
			"build.sslflags":"",
			"build.stacksmash_flags":"",
			"build.stdcpp_level":"-std=gnu++17",
			"build.stdcpp_lib":"-lstdc++",
			"build.system.path":"C:\Users\Niko\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\system",
			"build.variant":"nodemcu",
			"build.variant.path":"C:\Users\Niko\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\variants\nodemcu",
			"build.vtable_flags":"-DVTABLES_IN_FLASH",
			"commonhfile.fqfn":"C:\Users\Niko\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/CommonHFile.h",
			"compiler.S.cmd":"xtensa-lx106-elf-gcc",
			"compiler.S.extra_flags":"",
			"compiler.S.flags":"-c -g -x assembler-with-cpp -MMD -mlongcalls "-IC:\Users\Niko\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/include/"",
			"compiler.ar.cmd":"xtensa-lx106-elf-ar",
			"compiler.ar.extra_flags":"",
			"compiler.ar.flags":"cru",
			"compiler.as.cmd":"xtensa-lx106-elf-as",
			"compiler.c.cmd":"xtensa-lx106-elf-gcc",
[... 1336 lines cut ...]
		"version":"3.1.0-gcc10.3-e5f9fec"},
		{"name":"python3",
		"packager":"esp8266",
		"version":"3.7.2-post1"},
		{"name":"xtensa-lx106-elf-gcc",
		"packager":"esp8266",
		"version":"3.1.0-gcc10.3-e5f9fec"}
		]
	},
	"port":{"label":"COM10",
	"address":"COM10",
	"hardwareId":"",
	"properties":{"pid":"0x7523",
	"serialNumber":"",
	"vid":"0x1A86"},"protocol":"serial",
	"protocolLabel":"Serial Port (USB)"},"userDirPath":"C:\Users\Niko\Documents\Arduino",
	"dataDirPath":"C:\Users\Niko\AppData\Local\Arduino15"
}
```

**and finally I have error:**
LittleFS Filesystem Uploader
ERROR: No data folder found

so think will be better if you add text in error like "If recompile not helps - try to replace board library" or you can verify JSON nodes from my first post and show error "Used unsupported board library".

kittaakos, 2023-11-23:
@boomsya, since you're using version `2.2.1`, you might see an IDE bug (https://github.com/arduino/arduino-ide/pull/2233) manifesting in this extension. There is a very similar issue in another Arduino IDE extension: https://github.com/dankeboy36/esp-exception-decoder/issues/10#issuecomment-1737894487.

Please try if you can reproduce the bug with the [nightly IDE2 build](https://www.arduino.cc/en/software#nightly-builds). There may be no problem with this extension.

boomsya, 2023-11-23:
thanks. I will try, but now I have success upload LittleFS after rename folder "website" to "data".
[image]
think text "No data folder found" you can replace with "No data folder found. Check if the folder "data" exists is sketch folder"

closed 2023-11-23 by boomsya

boomsya, 2023-11-23:
after update IDE to 2.2.2-nightly have `ERROR: No data folder found` but project structure is not modified after last success upload. A new challenge accepted :)

Venky5451, 2024-11-07:
Restarting the IDE resolves the error for
`Board details not found`