earlephilhower/arduino-littlefs-upload #6: Installation Instructions - Windows 10

candideu, 2023-12-03:
Hi there! Where can I find `~/.arduinoIDE/plugins/` in Windows 10?

Is it this?

[image]

Or the Sketch location?

[image]

candideu, 2023-12-04:
[Correction](https://github.com/earlephilhower/arduino-littlefs-upload/pull/7#pullrequestreview-1761411753): 

`C:\Users\<username>\.arduinoIDE\`

simooon442, 2024-01-22:
Hello.
Do we have to put in this folder only **arduino-littlefs-upload-1.0.0.vsix** file ? Or the entire shared folder ?
I tried to put put both, but unable to see littlefs in the IDE tools bar

boomsya, 2024-01-22:
Usage
[Ctrl] + [Shift] + [P], then "Upload LittleFS to Pico/ESP8266".

simooon442, 2024-01-22:
1- I copied the VSIX file on the /.arduinoIDE/plugins/ folder
2- I pressed shift+P to open the arduino CLI, then copy and validate "upload Little FS..."

But I get error "Board details not available. Compile the sketch once"

Even if I compile before doing instructions

earlephilhower, 2024-01-22:
See https://github.com/earlephilhower/arduino-littlefs-upload#glitches

simooon442, 2024-01-23:
So it works fine ! thanks !