earlephilhower/arduino-littlefs-upload #9: Port is busy during file system upload

abaymar, 2023-12-10:
I cannot upload files as Arduino IDE complains that the port is busy and gives a fatal error:

_A fatal esptool.py error occurred: [Errno 16] could not open port /dev/cu.usbserial-0001: [Errno 16] Resource busy: '/dev/cu.usbserial-0001'_

The only solution I found is this:

- Select an incorrect port, for example: "/dev/cu.wlan-debug" in Arduino IDE Tools/Port
- Start sketch uploading and wait until the upload procedure becomes blocked, but not until it times out
- Select the correct port (that is the one with the ESP attached, in my case: "/dev/cu.usbserial-0001") in Arduino IDE Tools/Port
- Upload the files
- Upload the sketch if needed

MacBook Air M1
Arduino 2.2.1
ESP8266 NodeMCU 1.0

earlephilhower, 2023-12-10:
Close the Serial Monitor and try again.  There is no way for a plug in to do that for you per the Arduino folks.

closed 2023-12-10 by earlephilhower

abaymar, 2023-12-10:
Indeed closing the Serial Monitor helps!