earlephilhower/arduino-littlefs-upload #12: ERROR: No port specified, check IDE menus.  - although serial monitor connects successfully

slowjoe007, 2024-01-14:
Hi @earlephilhower, thank you for the work on this plugin. I did use it successfully before. Currently, however, I am facing issues connecting to the board. The error message does not give me a clue.
```text
LittleFS Filesystem Uploader

ERROR: No port specified, check IDE menus.
```
The port (COM7) is specified in "Tools > Port" in the Arduino IDE. And in fact Serial Monitor can connect and display messages successfully. (For the upload attempts I detached the Serial Monitor as I read it cannot be used concurrently.) This should show that the port is specified correctly and works.  

What else could be checked? Which (other) IDE menu might be referred to by the error message? Maybe the error message can be enhanced.  

Thank you in advance for your assistance.
-slowjoe

---
Environment:
- Windows 10
- Arduino IDE
  - Version: 2.2.1
  - Date: 2023-08-31T14:35:44.802Z
  - CLI Version: 0.34.0
- Board: LOLIN(WEMOS) D1 mini (clone)

boomsya, 2024-01-14:
the same problem. Helps only restart IDE and try again

dacarson, 2024-04-19:
I see this occasionally too. It happens when I have two IDE windows open and I am switching between the two. For some reason, the IDE stops sharing the port with the extension. 
When this happens, I need to close all IDE windows, and open the only one I want to do the upload with. Then it works.

bastianrickmannACN, 2024-05-14:
why would this be closed as an issue, when this is clearly a bug which isn't fixed? I mean at least the error would need to be catched correly and then tell the actual issue.

Oleks0ir, 2024-10-27:
Same here. Now I needed to reload my whole PC for it to finally work, as the bug was still present even after reinstalling LittleFS

1e1, 2024-11-11:
Still there on LittleFS Filesystem Uploader v1.3.0 and Arduino IDE 2.3.3
Restarting the IDE worked for a while. 
Now, even restarting the computer does not work.

sanjazahn, 2025-01-23:
You have to close monitor

Vinay0827, 2025-02-11:
I'm also facing the same problem with the Windows system but it's working fine in  my Ubuntu system what should I do

cacaodev, 2025-05-05:
Same here with macos 15.4.1, Arduino IDE 2.3.5 and Serial Monitor closed.

craynernh, 2025-08-24:
8/25 - windows = still broken - moving to linux Arduino IDE 2.3.5 and Serial Monitor closed.

1e1, 2025-08-27:
Not sure about the fix: close the 'Serial Monitor' tab.

gabalfonso, 2026-11-08:
I ran into `ERROR: No port specified, check IDE menus.` on Windows with an ESP32-S3. The Arduino IDE showed COM3, and the Serial Monitor could connect. Restarting the IDE and reselecting COM3 from **Tools → Port** didn’t help.

Here’s what worked for me:

1. Press Ctrl + Shift + P to open the Command Palette.
2. Run **`Developer: Arduino: Clear the board and port selection`**.
3. Use the board selector to select your board and port again. In my case, these were ESP32S3 Dev Module and COM3.
4. Open the Command Palette again, search for **`Upload LittleFS`**, and run the matching command. In my IDE, it was labeled `Upload LittleFS to Pico/ESP8266/ESP32`.

COM3 was my only available port. Clearing the selection and choosing the same port again let the upload complete.

Since the developer command clears the Arduino IDE’s board and port selection, this workaround is worth trying on other supported boards when the IDE shows the correct port but LittleFS reports this error.

This might be worth adding to the Glitches section of the README.