earlephilhower/arduino-littlefs-upload #15: ESP32 support

dacarson, 2024-04-18:
Please add ESP32 support. 

It looks like it could be added similar to its predecessor, ESP8266. The command line tool is there on my mac:
./Library/Arduino15/packages/esp32/tools/mklittlefs
./Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933/mklittlefs

I'd be willing to work on this myself, if you are able to provide instructions on how to build the VSIX file. I see that it is a deflate/zip file, but I am not sure how to build it.

earlephilhower, 2024-04-18:
The problem isn't making the filesystem, unfortunately (in fact, I think they use my mklittlefs tool or a close fork).  The issue is the ESP32 partitioning infrastructure.  I never really got into the ESP32 ecosystem and am unfamiliar with how they go about doing things like FS partitions/etc.

That said, a PR adding ESP32 support would be much appreciated!

My build-and-test.sh script simply calls
````
rm -f *.vsix
vsce package
````
to build the bundle.  To remove old versions, build and install the edited one, and test it out I use:
````
#!/bin/bash
rm -f *.vsix; vsce package && ( rm -rf ~/.arduinoIDE/plugins/* /tmp/vscode-unpacked/ ; cp *vsix ~/.arduinoIDE/plugins/. ) && ~/Downloads/arduino-ide_nightly-20230925_Linux_64bit.AppImage
````

dacarson, 2024-04-19:
For those following along, to build the project I needed to install a couple of npm packages:
`npm install -g vsce`
`npm install -g typescript`
`npm install` (post install script to fetch vscode libraries?)
`npm install -g vscode-arduino-api`
Lastly in `tsconfig.json`, add under `compilerOptions`  `"allowSyntheticDefaultImports": true` 

To add support for ESP32 I examined this Java project: [arduino-esp32littlefs-plugin](https://github.com/lorol/arduino-esp32littlefs-plugin). 
The partitions offered in the menu map to CSV files. Parsing the CSV files, you can find the spiffs entry. 

I was able to translate it from Java to Typescript, and merge it into your project. 
Patch coming.

earlephilhower, 2024-04-19:
Great, thanks.  I think I followed some online VSCode plug-in development blog to get those `npm` packages installed so your explicit listing is very handy.  You may want to edit the `readme.md` file, too, to add those bits in for posterity if you have a second.

One thing, though, my tsconfig has no `allowSynthetic...` in it so maybe that's not really required (or maybe for the CSV/JSON parsing?):
````
$ cat tsconfig.json 
{
	"compilerOptions": {
		"module": "commonjs",
		"target": "ES2020",
		"outDir": "out",
		"lib": [
			"ES2020"
		],
		"sourceMap": true,
		"rootDir": "src",
		"strict": true   /* enable all strict type-checking options */
		/* Additional Checks */
		// "noImplicitReturns": true, /* Report error when not all code paths in function return a value. */
		// "noFallthroughCasesInSwitch": true, /* Report errors for fallthrough cases in switch statement. */
		// "noUnusedParameters": true,  /* Report errors on unused parameters. */
	}
}
````

dacarson, 2024-04-19:
Without the change in `tsconfig.json`, I get a stream of errors like:
```
node_modules/ardunno-cli/dist/api/google/rpc/status.d.ts:1:8 - error TS1192: Module '"/Users/dacarson/Documents/GitHub/arduino-littlefs-upload/node_modules/protobufjs/minimal"' has no default export.

1 import _m0 from 'protobufjs/minimal';
```
and it fails to compile:
```
Found 14 errors in 14 files.

Errors  Files
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/board.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/commands.d.ts:2
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/common.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/compile.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/core.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/debug.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/lib.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/monitor.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/port.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/commands/v1/upload.d.ts:1
     1  node_modules/ardunno-cli/dist/api/cc/arduino/cli/settings/v1/settings.d.ts:2
     1  node_modules/ardunno-cli/dist/api/google/protobuf/any.d.ts:1
     1  node_modules/ardunno-cli/dist/api/google/protobuf/wrappers.d.ts:1
     1  node_modules/ardunno-cli/dist/api/google/rpc/status.d.ts:1
 ERROR  npm failed with exit code 2
```

Googling the issue, it can be fixed with the change to the `tsconfig.json` file

dacarson, 2024-04-19:
FWIW, I did not add the changes to `tsconfig.json` to the PR.

Would be interest to know if it compiles fine for you.

closed 2024-04-19 by earlephilhower

dacarson, 2024-04-19:
It would be good to update the [release](https://github.com/earlephilhower/arduino-littlefs-upload/releases) to 1.1.0

Also update the project 'About' to include ESP32:-
"Build and uploads LittleFS filesystems for the Arduino-Pico RP2040 and ESP8266 cores under Arduino IDE 2.2.1 or higher."

earlephilhower, 2024-04-19:
Added the ESP32 to the about statement.

For some reason when I did the 1.1.0 release it wasn't set as the latest.  Corrected and now shows up on the landing page.