earlephilhower/arduino-littlefs-upload #18: Fatal upload error, Chip is ESP32-C3 not ESP32

buccaneer-jak, 2024-04-23:
Hi, I received this error while uploading to an ESP32-C3. Is the C3 not supported yet as I thought it may be covered under the ESP32 cores?

Uploading LittleFS filesystem
C:\Users\jak\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe --chip esp32 --port COM6 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 C:\Users\jak\AppData\Local\Temp\tmp-36716-6YiG3mVg3Z5z-.littlefs.bin
esptool.py v4.5.1
Serial port COM6
Connecting...................

A fatal error occurred: This chip is ESP32-C3 not ESP32. Wrong --chip argument?
ERROR:  Upload failed, error code: 2

earlephilhower, 2024-04-23:
Yeah, it looks like the `--chip` is hardcoded but should come from the build environment.  Would you be able to capture the upload `esptool.py` command line from IDE 1.8 for a successful upload and post it?  I'll look through the Arduino board info values in the meantime.

earlephilhower, 2024-04-23:
@buccaneer-jak can you try the [new 1.1.1 binary in the PR here](https://github.com/earlephilhower/arduino-littlefs-upload/raw/eed0547db3a3d33bb37f39f25278c7effa4af315/arduino-littlefs-upload-1.1.1.vsix)?  Remove the 1.1.0 file and install this version and re-run.  It uses the FQBN to find the proper `--chip` argument.

earlephilhower, 2024-04-23:
I'm also a little suspect as to the actual `esptool` call in the plug in for the ESP32.  Again, I'm not really familiar with the ESP32 infrastructure but it seems like it's calling a `.py` file and assuming the shell/`bash`/`Powershell` will find the right app vs. explicitly calling the `python` interpreter...

buccaneer-jak, 2024-04-23:
Eureka!, LittleFS upload succeeded.

LittleFS Filesystem Uploader

Using partition: default
Building LittleFS filesystem
C:\Users\jak\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933/mklittlefs.exe -c C:\Users\jak\OneDrive\Documents\Arduino\Sketches\Testing\INA3221\ESP32-C3_INA3221/data -p 256 -b 4096 -s 1441792 C:\Users\jak\AppData\Local\Temp\tmp-30868-IzayIpjkzkIO-.littlefs.bin

Uploading LittleFS filesystem
C:\Users\jak\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe --chip esp32c3 --port COM6 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 C:\Users\jak\AppData\Local\Temp\tmp-30868-IzayIpjkzkIO-.littlefs.bin
esptool.py v4.5.1
Serial port COM6
Connecting...........................
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 64:e8:33:03:af:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00290000 to 0x003effff...
Compressed 1441792 bytes to 1501...
Writing at 0x00290000... (100 %)
Wrote 1441792 bytes (1501 compressed) at 0x00290000 in 11.9 seconds (effective 966.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Completed upload.

Thanks, works a treat. I will be testing the ESP32-S3 in a few days with 16mB and will update you.

closed 2024-04-23 by earlephilhower

eyaler, 2025-05-12:
@buccaneer-jak were you able to get it to work with ESP32-S3 16MB? if so could you share the partition csv? thanks!

commit eed0547d (2024-04-23): Use custom ESP32 board as defined in FQBN

Fixes #18
PR #19 (2024-04-23): Use custom ESP32 board as defined in FQBN
Fixes #18
commit f0f51402 (2024-04-23): Use custom ESP32 board as defined in FQBN (#19)

Fixes #18