earlephilhower/arduino-littlefs-upload #25: esptool -esp chip option wrong

alka79, 2024-04-28:
Arduino IDE 2.3.2 ,  windows 10 , MH-ET-LIVE ESP32MINIKit board.

esptool complains about --chip option

```Uploading LittleFS filesystem
C:\Users\me\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe --chip mhetesp32minikit --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 C:\Users\me\AppData\Local\Temp\tmp-15956-W0ZrEjg5ggbn-.littlefs.bin
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2,esp32c6,esp32h2}]
               [--port PORT] [--baud BAUD]
```

When I run the above command from a shell with `--chip esp32` the upload works fine.

earlephilhower, 2024-04-28:
What is your exact FQBN?  It looks like you are not using the official ESP32 core, which has "esp32" as the architecture (2nd section of the FQBN).

alka79, 2024-04-29:
My board is branded by AZ Deliverly. It is a clone of  Wemos MINI D1 ESP32 also known as MH ET LIVE MINIKIT ESP32. Hard to know which one is the original manufacturer !

I tried with both boards names
- d1_mini32
- mhetesp32minikit

littlefs uploads fails with both for the same reason : --chip option incorrect. It seems that --chip option passed to esptool_py is the board FQBN name instead of the architecture.

I use latest boards ESP32 2.0.15

this is the description in board.txt of one of the boards
##############################################################

mhetesp32minikit.name=MH ET LIVE ESP32MiniKit

mhetesp32minikit.bootloader.tool=esptool_py
mhetesp32minikit.bootloader.tool.default=esptool_py

mhetesp32minikit.upload.tool=esptool_py
mhetesp32minikit.upload.tool.default=esptool_py
mhetesp32minikit.upload.tool.network=esp_ota

mhetesp32minikit.upload.maximum_size=1310720
mhetesp32minikit.upload.maximum_data_size=327680
mhetesp32minikit.upload.flags=
mhetesp32minikit.upload.extra_flags=

mhetesp32minikit.serial.disableDTR=true
mhetesp32minikit.serial.disableRTS=true

mhetesp32minikit.build.tarch=xtensa
mhetesp32minikit.build.bootloader_addr=0x1000
mhetesp32minikit.build.target=esp32
mhetesp32minikit.build.mcu=esp32
mhetesp32minikit.build.core=esp32
mhetesp32minikit.build.variant=mhetesp32minikit
mhetesp32minikit.build.board=MH_ET_LIVE_ESP32MINIKIT

mhetesp32minikit.build.f_cpu=240000000L
mhetesp32minikit.build.flash_mode=dio
mhetesp32minikit.build.flash_size=4MB
mhetesp32minikit.build.boot=dio
mhetesp32minikit.build.partitions=default
mhetesp32minikit.build.defines=

mhetesp32minikit.menu.FlashFreq.80=80MHz
mhetesp32minikit.menu.FlashFreq.80.build.flash_freq=80m
mhetesp32minikit.menu.FlashFreq.40=40MHz
mhetesp32minikit.menu.FlashFreq.40.build.flash_freq=40m

mhetesp32minikit.menu.PartitionScheme.default=Default with spiffs
mhetesp32minikit.menu.PartitionScheme.default.build.partitions=default
mhetesp32minikit.menu.PartitionScheme.defaultffat=Default with ffat
mhetesp32minikit.menu.PartitionScheme.defaultffat.build.partitions=default_ffat
mhetesp32minikit.menu.PartitionScheme.no_ota=No OTA (Large APP)
mhetesp32minikit.menu.PartitionScheme.no_ota.build.partitions=no_ota
mhetesp32minikit.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
mhetesp32minikit.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (Large APPS with OTA)
mhetesp32minikit.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
mhetesp32minikit.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080

mhetesp32minikit.menu.UploadSpeed.921600=921600
mhetesp32minikit.menu.UploadSpeed.921600.upload.speed=921600
mhetesp32minikit.menu.UploadSpeed.115200=115200
mhetesp32minikit.menu.UploadSpeed.115200.upload.speed=115200
mhetesp32minikit.menu.UploadSpeed.256000.windows=256000
mhetesp32minikit.menu.UploadSpeed.256000.upload.speed=256000
mhetesp32minikit.menu.UploadSpeed.230400.windows.upload.speed=256000
mhetesp32minikit.menu.UploadSpeed.230400=230400
mhetesp32minikit.menu.UploadSpeed.230400.upload.speed=230400
mhetesp32minikit.menu.UploadSpeed.460800.linux=460800
mhetesp32minikit.menu.UploadSpeed.460800.macosx=460800
mhetesp32minikit.menu.UploadSpeed.460800.upload.speed=460800
mhetesp32minikit.menu.UploadSpeed.512000.windows=512000
mhetesp32minikit.menu.UploadSpeed.512000.upload.speed=512000

mhetesp32minikit.menu.DebugLevel.none=None
mhetesp32minikit.menu.DebugLevel.none.build.code_debug=0
mhetesp32minikit.menu.DebugLevel.error=Error
mhetesp32minikit.menu.DebugLevel.error.build.code_debug=1
mhetesp32minikit.menu.DebugLevel.warn=Warn
mhetesp32minikit.menu.DebugLevel.warn.build.code_debug=2
mhetesp32minikit.menu.DebugLevel.info=Info
mhetesp32minikit.menu.DebugLevel.info.build.code_debug=3
mhetesp32minikit.menu.DebugLevel.debug=Debug
mhetesp32minikit.menu.DebugLevel.debug.build.code_debug=4
mhetesp32minikit.menu.DebugLevel.verbose=Verbose
mhetesp32minikit.menu.DebugLevel.verbose.build.code_debug=5

mhetesp32minikit.menu.EraseFlash.none=Disabled
mhetesp32minikit.menu.EraseFlash.none.upload.erase_cmd=
mhetesp32minikit.menu.EraseFlash.all=Enabled
mhetesp32minikit.menu.EraseFlash.all.upload.erase_cmd=-e

##############################################################

earlephilhower, 2024-04-29:
Looks like the ESP32 doesn't really use the FQBN architecture since there's only 1 core but something like 6 variants.  Please try #26 and report back, it uses the `build.mcu` setting which seems to match their `platform.txt` build recipes.

alka79, 2024-04-29:
1.1.5 works fine for me on ESP32. Thanks a lot for your work.

PR #26 (2024-04-29): Use build.mcu, not FQBN archtecture, for ESP32
Fix #25
commit ff27d47d (2024-04-29): Use build.mcu, not FQBN archtecture, for ESP32 (#26)

Fix #25
closed 2024-04-29 by earlephilhower