earlephilhower/arduino-littlefs-upload #27: File system full

candideu, 2024-05-02:
Hi! I'm starting to get the following error, using the version released three days ago (1.1.5):

```html
LittleFS Filesystem Uploader

Building LittleFS filesystem
C:\Users\[user]\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec/mklittlefs.exe -c C:\Users\[user]\Documents\Arduino\LittleFS-test/data -p 256 -b 8192 -s 2072576 C:\Users\[user]\AppData\Local\Temp\tmp-9752-JpwJrT6euN4a-.littlefs.bin
/colouringbook.js
/index.html
littlefs/lfs.c:561:error: No more free space 371
lfs_write error(-28): File system is full.

error adding file!
ERROR:  Mklittlefs failed, error code: 1
```

My sketch + data folder are 3MB total, and I'm using a WEMOS D1 Mini Pro (16MB Flash Memory)

earlephilhower, 2024-05-02:
Your FS is set up to only be 2MB (`-s 2072576`)and you're trying to upload 3MB of data, so it's not going to work too well.  Increase the FS size through the menus and try again.

closed 2024-05-02 by earlephilhower

candideu, 2024-05-02:
Thank you @earlephilhower !