earlephilhower/arduino-littlefs-upload #31: No data folder found

markingle, 2024-05-15:
I am using an ESP32 Dev Module running Arduino 2.3.2.  I have the following data directory with my sketch

C:\Users\marka\Documents\Arduino\Aerator_Livewell_Switch\data

earlephilhower, 2024-05-15:
This is an IDE glitch, nothing to do here.  See the [workaround for the serial port,](https://github.com/earlephilhower/arduino-littlefs-upload?tab=readme-ov-file#glitches) the same should work here.

closed 2024-05-15 by earlephilhower

markingle, 2024-05-15:
Hello!  The fix suggested does not work for me.  The plugin still indicates a missing data folder

markingle, 2024-05-15:
[image]

[image]

earlephilhower, 2024-05-15:
Try building the sketch once before doing the upload.  Either the IDE isn't reporting the proper value for the sketch directory, or it doesn't have permission to access files inside it (i.e. I've heard that OneDrive folders are sometimes weird that way).

earlephilhower, 2024-05-15:
You can also try the more verbose version I just threw together which *may* shed some light on your particular problem.  Remove the 1.1.5 and download and install the 1.1.6 version and restart things and re-run.

https://github.com/earlephilhower/arduino-littlefs-upload/raw/talkie/arduino-littlefs-upload-1.1.6.vsix

markingle, 2024-05-15:
Thank you!  I will give it a try now

markingle, 2024-05-15:
Thank you for taking the time to provide some logging out put!  I had embedded the sketch directory within itself some how.....It could only be seen in the logging.

[image]

markingle, 2024-05-15:
I recommend keeping the log output...
[image]

earlephilhower, 2024-05-16:
Great, thanks for the follow up!  I've made that new version the release one and it'll hopefully make everyone's life easier.