earlephilhower/arduino-littlefs-upload #35: ERROR:  Upload failed, error code: 1

mesh-newbie, 2024-05-24:
Hi,
Issue when uploading to the pico-w :-(
Hope you can help.

`
LittleFS Filesystem Uploader v1.1.7 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: /Users/craig/Documents/Arduino/rpicow
   Data Path: /Users/craig/Documents/Arduino/rpicow/data
      Device: RP2040 series

Building LittleFS filesystem
Command Line: /Users/craig/Library/Arduino15/packages/rp2040/tools/pqt-mklittlefs/2.2.0-d04e724/mklittlefs -c /Users/craig/Documents/Arduino/rpicow/data -p 256 -b 4096 -s 65536 /var/folders/dh/qd12r7d170sfs_6nhs13fsnh0000gn/T/tmp-18226-NK5yQ5PRGw7n-.littlefs.bin
/update.html
/index.html
/styles.css
/scripts.js

Uploading LittleFS filesystem
Command Line: /Users/craig/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3 /Users/craig/Library/Arduino15/packages/rp2040/hardware/rp2040/3.8.1/tools/uf2conv.py --base 270462976 --serial /dev/cu.usbmodem11312201 --family RP2040 /var/folders/dh/qd12r7d170sfs_6nhs13fsnh0000gn/T/tmp-18226-NK5yQ5PRGw7n-.littlefs.bin
Resetting /dev/cu.usbmodem11312201
Converting to uf2, output size: 131072, start address: 0x101ef000
Scanning for RP2040 devices
No drive to deploy.

ERROR:  Upload failed, error code: 1

`
Arduino IDE 2.3.2 on Mini Mac M1, 2020
Flash Size: 2MB "(Sketch: 1984KB, FS: 64KB)" (have tried other configs as well, still same)
Contents of /data is < 1KB

I can upload normally using the IDE no problem. 

FYI
`
#include <WiFi.h>
#include <WebServer.h>
#include <LittleFS.h>
#include <ArduinoOTA.h>
#include <NTPClient.h>
#include <WiFiUdp.h>
#include "credentials.h"
#include "hardware.h"
#include "webpages.h"
.. // rest of code
`

earlephilhower, 2024-05-24:
First off, make sure you have closed and and all serial monitors before uploading.  The plugin cannot do this automatically and they can't both talk to the port.  Also, if the Pico has crashed or hung then the USB port won't work for upload since the Pico needs code to run the port.

The FS uploader uses the same tool as the main IDE upload, so the usual "hold bootsel while plugging in" to get into bootloader mode will work.

closed 2024-05-24 by earlephilhower

mesh-newbie, 2024-05-24:
> First off, make sure you have closed and and all serial monitors before uploading. The plugin cannot do this automatically and they can't both talk to the port. Also, if the Pico has crashed or hung [...]

Many thanks.

Yup. That was the trick. Closing the Serial Monitor :-)
Perhaps add note in ReadMe for newbies like me.

MicSG-dev, 2024-05-25:
@mesh-newbie 
> Yup. That was the trick. Closing the Serial Monitor :-) Perhaps add note in ReadMe for newbies like me.

[image]