earlephilhower/arduino-littlefs-upload #36: Suggestion: list of plugins compatible with Arduino IDE 2.0

MicSG-dev, 2024-05-25:
As a suggestion, there could be a list in some repository with plugins compatible with the Arduino IDE 2.0.
The ones I know are:
- [arduino-littlefs-upload](https://github.com/earlephilhower/arduino-littlefs-upload): LittleFS loader;
- [WakaTime for Arduino IDE](https://wakatime.com/arduino-ide): Fully automatic time control to assist with time management in each project.

earlephilhower, 2024-05-25:
Why not start a forum post at [Arduino](https://forum.arduino.cc/) or make a repo yourself, like they do for the [Arduino Libraries](https://github.com/arduino/library-registry)?

MicSG-dev, 2024-05-26:
Forum post created: https://forum.arduino.cc/t/list-of-plugins-compatible-with-arduino-ide-2-0
@earlephilhower Thanks for the suggestion.

closed 2024-05-26 by MicSG-dev