earlephilhower/arduino-littlefs-upload #40: ERROR: Partition file not found!

desy0305, 2024-06-12:
**Partition Scheme SPIFFS:**
LittleFS Filesystem Uploader v1.1.8 -- https://github.com/earlephilhower/arduino-littlefs-upload
 Sketch Path: C:\Users\USER\Documents\Arduino\673OEE_db
   Data Path: C:\Users\USER\Documents\Arduino\673OEE_db\data
      Device: ESP32 series, model esp32s3
Using partition: spiffs
  Partitions: C:\Users\USER\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\tools\partitions\spiffs.csv
ERROR: Partition file not found!

**Partition Scheme Default:**
LittleFS Filesystem Uploader v1.1.8 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: C:\Users\USER\Documents\Arduino\673OEE_db
   Data Path: C:\Users\USER\Documents\Arduino\673OEE_db\data
      Device: ESP32 series, model esp32s3
Using partition: default
  Partitions: C:\Users\USER\AppData\Local\Arduino15\packages\arduino\hardware\esp32\2.0.13\tools\partitions\default.csv

Building LittleFS filesystem
Command Line: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933\mklittlefs.exe -c C:\Users\LMATEEV\Documents\Arduino\673OEE_db\data -p 256 -b 4096 -s 1441792 C:\Users\LMATEEV\AppData\Local\Temp\tmp-58792-b8pjMUPqCff5-.littlefs.bin
/data.sqlite3

Uploading LittleFS filesystem
Command Line: C:\Users\USER\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1\esptool.exe --chip esp32s3 --port COM12 --baud NaN --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 C:\Users\USER\AppData\Local\Temp\tmp-58792-b8pjMUPqCff5-.littlefs.bin
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2,esp32c6,esp32h2}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,get_security_info,version}
               ...
esptool: error: argument --baud/-b: invalid arg_auto_int value: 'NaN'
ERROR:  Upload failed, error code: 2

earlephilhower, 2024-06-12:
Your configuration looks like it has several  problems, and it's several versions behind on the Espressif Arduino core.  The error is saying first of all that there was no partition file in the standard spot for, and that the IDE doesn't have an upload speed defined for the board, which is needed by esptool even for app uploads.

I suggest moving to the 3.0.1 release, verifying app upload works, and trying again.  If that still fails, I'll need all the menu items selected (a screen cap of the menu would do) to see if it's reproducible outside of your setup.

desy0305, 2024-06-13:
Hi, maybe be becasue i was trying Spiffs and then trying littlefs.    I installed latest tools and plugins found. Seriam Monitor is closed, Programmer: Esptoll.   Can you give me hints what exactly you mean to be updated?
Screenshots:
[image]
[image]

Programmer: Esptoll

if  app3M_spiffs9M_fact512k_16MB.csv  is renamed to  spiffs.csv 
[image]

earlephilhower, 2024-06-13:
DFU upload is not supported, only the standard serial, which is the root of your problem w/esptool.  Only serial upload is available here.

closed 2024-06-13 by earlephilhower

desy0305, 2024-06-13:
I just tried all clickable options to illustrate that none of them worked. Tried spliffs,... No partition or missing BAud is always what I get, tried code to format file system and build but also no success. I am newbie but already spent 2 full days trying to get simple .sqlite upload into the board.   Tried to add latest esptool.exe also no success. So problem is still active

earlephilhower, 2024-06-13:
Again, the problem is that board, unlike 99% of the other ESP32 boards, doesn't support esptool upload thru the serial port.  Only DFU is enabled which breaks esptool and is not supported here.

desy0305, 2024-06-13:
Thank you, now much clearer