earlephilhower/arduino-littlefs-upload #50: Custom partition file not handled correctly

FrenkK, 2024-09-11:
I'm trying to get every last byte out of the 4 MB ESP32 I'm using. I have a smallish app with a bunch of image files I want to put in the SPIFFS and wanted to build a custom partition setup.

I use the partitions.csv file in my sketch directory:

```
# Name,   Type, SubType, Offset,  Size, Flags
factory,app,factory,0x10000,0x80000,
spiff,data,spiffs,0x90000,0x370000,
```

I put the files in the data folder as instructed and when I run the upload tool, I get the "ERROR: Partition entry not found in csv file!".

If I choose one of the predefined partition setups, the tool works.

closed 2024-09-11 by earlephilhower

FrenkK, 2024-09-12:
@earlephilhower Thank you for your response, but that made no difference. The first column is the name anyway, probably shouldn't matter.

earlephilhower, 2024-09-12:
Actually, it does key off of the 1st entry (name).  I don't use ESP32, but the contributor who added this chose that.

And I just tried your example and failed:
````
$ cat /home/earle/Arduino/Blink_copy_20240608074928/partitions.csv
# Name,   Type, SubType, Offset,  Size, Flags
factory,app,factory,0x10000,0x80000,
spiff,data,spiffs,0x90000,0x370000,
````
````
LittleFS Filesystem Uploader v1.1.9 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: /home/earle/Arduino/Blink_copy_20240608074928
   Data Path: /home/earle/Arduino/Blink_copy_20240608074928/data
      Device: ESP32 series, model esp32c3
Using partition: partitions.csv in sketch folder
  Partitions: /home/earle/Arduino/Blink_copy_20240608074928/partitions.csv

ERROR: Partition entry not found in csv file!
````

BUT, fixing the name to "spiffs" makes it work fine:
````
$ cat /home/earle/Arduino/Blink_copy_20240608074928/partitions.csv
# Name,   Type, SubType, Offset,  Size, Flags
factory,app,factory,0x10000,0x80000,
spiffs,data,spiffs,0x90000,0x370000,
````
````
LittleFS Filesystem Uploader v1.1.9 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: /home/earle/Arduino/Blink_copy_20240608074928
   Data Path: /home/earle/Arduino/Blink_copy_20240608074928/data
      Device: ESP32 series, model esp32c3
Using partition: partitions.csv in sketch folder
  Partitions: /home/earle/Arduino/Blink_copy_20240608074928/partitions.csv

Building LittleFS filesystem
Command Line: /home/earle/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933/mklittlefs -c /home/earle/Arduino/Blink_copy_20240608074928/data -p 256 -b 4096 -s 3604480 /tmp/tmp-224603-Z2vzslsDLD3o-.littlefs.bin
/info.txt

Uploading LittleFS filesystem
Command Line: python3 /home/earle/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool.py --chip esp32c3 --port /dev/ttyACM2 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 589824 /tmp/tmp-224603-Z2vzslsDLD3o-.littlefs.bin
esptool.py v4.6
Serial port /dev/ttyACM2
....
````

So, please check the name again is "spiffs" and try...

FrenkK, 2024-09-12:
@earlephilhower, like I said, I tried your suggestion:
```
# Name,   Type, SubType, Offset,  Size, Flags
factory,app,factory,0x10000,0x80000,
spiffs,data,spiffs,0x90000,0x370000,
````
The output is:
````
LittleFS Filesystem Uploader v1.1.8 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: D:\Moje stvari\Dokumenti\aaMail\Druge stvari\arduino\sketches\GxEPD2_1_bit_bitmaps_from_zipped_SPIFFS_480x800
   Data Path: D:\Moje stvari\Dokumenti\aaMail\Druge stvari\arduino\sketches\GxEPD2_1_bit_bitmaps_from_zipped_SPIFFS_480x800\data
      Device: ESP32 series, model esp32
Using partition: partitions.csv in sketch folder
  Partitions: D:\Moje stvari\Dokumenti\aaMail\Druge stvari\arduino\sketches\GxEPD2_1_bit_bitmaps_from_zipped_SPIFFS_480x800\partitions.csv

ERROR: No port specified, check IDE menus.
````

earlephilhower, 2024-09-12:
Err, you now have an upload problem completely unrelated to the original one.  Make sure the connection is set (only serial, no OTA), and the menus are selected correctly as it says.  You should be able to upload a sketch first before doing a FS upload.

FrenkK, 2024-09-12:
It works now, sorry for wasting your time. But I could swear it didn't the first time I tried.