earlephilhower/arduino-littlefs-upload #51: Pimoroni Pico 2 Plus: define TARGET_RP2040 missing and problem to upload data to flash using "Pico LittleFS Data Upload"

cledic, 2024-09-13:
Hello to all,
I am using a **Pimoroni Pico 2 Plus** with the "GFX_Library_for_Arduino" library ver 1.4.9, but the "ImgViewerMjpeg_JPEGDEC" example compiles only by inserting the define: "**TARGET_RP2040**".

I then inserted the command "-DTARGET_RP2040" among the other defines in the board.txt file and the compilation ended successfully. I know that it is not a viable solution, but: can you suggest me how I can insert it correctly in compilation?

I have another problem: **I can not upload the mjpeg files to flash.**
I followed the instructions and installed the picolittlefs.JAR file. When I run the "Data Upload", the procedure does not complete, the UF2 volume window on my PC remains open, and the communication is interrupted: I am forced to do a RESET to the board. After the restart, the program cannot find the file.

I have tried both test programs for the LittleFS software: "FSUpload" and "SpeedTest", and they both work! It seems that the board's flash is working!

This is the console output when I start the file upload:
`
[LittleFS] data    : C:\Users\me\Documents\Arduino\RP2350_ImgViewerMjpeg_ver0.01\data
[LittleFS] size    : 4096KB
[LittleFS] page    : 256
[LittleFS] block   : 4096
/earth.mjpeg
/earth128.mjpeg
[LittleFS] upload   : C:\Users\me\AppData\Local\Temp\arduino_build_857947/RP2350_ImgViewerMjpeg_ver0.01.mklittlefs.bin
[LittleFS] address  : 281010176
[LittleFS] swerial  : COM29
[LittleFS] python   : C:\Users\me\AppData\Local\Arduino15\packages\rp2040\tools\pqt-python3\1.0.1-base-3a57aed\python3.exe
[LittleFS] uploader : C:\Users\me\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2\tools\uf2conv.py

Resetting COM29
Converting to uf2, output size: 8388608, start address: 0x10bfe000
Scanning for RP2040 devices
Flashing D: (RP2350)
Wrote 8388608 bytes to D:/NEW.UF2
`
The volume window on my PC remains like this, open, and the board is locked:
[image]

How can I resolve this issue?

TIA
Best Regards
Clemente

earlephilhower, 2024-09-13:
New commands to make a RP2350 bootrom compatible upload:
````
mklittlefs -c /home/earle/Arduino/data1/data -p 256 -b 4096 -s 262144 /tmp/tmp-608036-azMU6Jzqg2Bc-.littlefs.bin
picotool  uf2 convert /tmp/tmp-608036-azMU6Jzqg2Bc-.littlefs.bin -t bin tmp.uf2 -o 0x10fbe000 --family data
uf2conv.py --serial /dev/ttyACM1 --family RP2040 tmp.uf2 --deploy
````

closed 2024-09-13 by earlephilhower

earlephilhower, 2024-09-13:
I'm not going to be updating the JAVA 1.x plug in as the 1.x IDE is EOL and last updated around 3 years ago.  If someone wants to backport this to the obsolete plugin, I'll be happy to look at PRs...

PR #52 (2024-09-13): Support RP2350 uploads automatically
The RP2350 needs special UF2 processing to support a binary flash.  Adjust the command lines depending on the build revision for the Pico core to accomodate.

Fixes #51
commit 78f328ab (2024-09-13): Support RP2350 uploads automatically (#52)

The RP2350 needs special UF2 processing to support a binary flash.  Adjust
the command lines depending on the build revision for the Pico core to accomodate.

Fixes #51