earlephilhower/arduino-littlefs-upload #56: A fatal esptool.py error occurred: could not open port 'COM4': PermissionError(13, '�ܾ����ʡ�', None, 5)

PPPZW233, 2024-10-12:
A fatal esptool.py error occurred: could not open port 'COM4': PermissionError(13, '�ܾ����ʡ�', None, 5)

PPPZW233, 2024-10-12:
LittleFS Filesystem Uploader v1.2.0 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: C:\Users\16276\OneDrive\桌面\3_4_1_SPIFFS_File_server
   Data Path: C:\Users\16276\OneDrive\桌面\3_4_1_SPIFFS_File_server\data
      Device: ESP8266 series

Building LittleFS filesystem
Command Line: C:\Users\16276\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec\mklittlefs.exe -c C:\Users\16276\OneDrive\桌面\3_4_1_SPIFFS_File_server\data -p 256 -b 8192 -s 1024000 C:\Users\16276\AppData\Local\Temp\tmp-4152-9Vsc20Eh9Ag6-.littlefs.bin
/clock.js
/favicon.ico
/img/taichi-maker.jpg
/index.html
/JavaScript.js
/main.css

Uploading LittleFS filesystem
Command Line: C:\Users\16276\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe C:\Users\16276\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\tools\upload.py --chip esp8266 --port COM4 --baud 115200 write_flash 3145728 C:\Users\16276\AppData\Local\Temp\tmp-4152-9Vsc20Eh9Ag6-.littlefs.bin
esptool.py v3.0
Serial port COM4

A fatal esptool.py error occurred: could not open port 'COM4': PermissionError(13, '�ܾ����ʡ�', None, 5)
Completed upload.

per1234, 2024-10-12:
Hi @PPPZW233. This error can be caused by having the [Arduino IDE **Serial Monitor**](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-monitor) or [**Serial Plotter**](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-plotter) open when you attempt to upload the filesystem.

Make sure you have both of them closed and then try using the extension again.

The **Serial Monitor** view is a tab in the bottom panel of the Arduino IDE window so you might not even notice it is open. Click the **X** icon on the "**Serial Monitor**" tab in the bottom panel of the Arduino IDE window to close it:

[image]

earlephilhower, 2024-10-12:
Thanks, @per1234 .  The serial monitor needs to manually closed before upload.

closed 2024-10-12 by earlephilhower