earlephilhower/arduino-littlefs-upload #59: Feature request: update boards.txt with custom partition app size

aIecxs, 2024-10-16:
referring to arduino/arduino-ide/issues/2526

you're reading the IDE preferences for partition scheme, so I assume you will be able to change preferences also.

If sketch directory contains custom `partitions.csv` the uploading of sketch failes, when default partition scheme is selected. it would be useful to automatically change the IDE settings to custom partition scheme each time the add-on detects a `partitions.csv` in sketch directory.

Furthermore, it would be useful to read the app partition size and update `PartitionScheme.custom.upload.maximum_size` (in `boards.txt` or wherever during runtime) with real app size, so the calculation in percentage is not misleading anymore.

I ask this as a feature of your add-on because neither the IDE nor the platform developers feel responsible for fixing this.

Of course it would maybe sufficient to run a script from `platform.txt` which updates `boards.txt` automatically, but the problem is these configurations are somehow cached in `%UserProfile%\AppData\Roaming\arduino-ide\Local Storage\leveldb` and not updated.

If you reject this feature request, kindly help me to read the preferences from script. I tried to extract JSON from leveldb but this only works for plain text in log file, I have no clue where in ldb binary the partition scheme preferences are stored.

aIecxs, 2024-10-16:
inviting @schmidtsamuelw to the discussion

earlephilhower, 2024-10-19:
Plugins can't change values or menu selections, only read them.  So there's nothing this plug in can do, sorry.

Look at the source in `extension.ts`for how to read values in your own extension code.

closed 2024-10-19 by earlephilhower