earlephilhower/arduino-littlefs-upload #61: have error with custom csv

stef-ladefense, 2024-11-02:
according to the Expressif doc, https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-guides/partition-tables.html#creating-custom-tables
a csv can contain spaces and short expressions, like 3M instead of 0x300000.
except that your extension no longer works in this case.

OK with : 
nvs,         data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,      app,  ota_0,   0x10000, 0x300000,
app1,      app,  ota_1,   0x310000,0x300000,
spiffs,     data, spiffs,  0x610000,0x9E0000,
coredump, data, coredump,0xFF0000,0x10000,

KO with : 
nvs,      data, nvs,      0x9000,  0x5000,
otadata,  data, ota,   0xe000,  0x2000,
app0,     app,  ota_0, 0x10000,  3M,
app1,     app,  ota_1,                 ,3M,
spiffs,   data, spiffs,                   ,0x9E0000,
coredump, data, coredump,     ,0x10000,

earlephilhower, 2024-11-02:
Ah, interesting.  TypeScript should handle spaces in `parseInt` just fine, but those magnitude modifiers definitely will go boom.
https://github.com/earlephilhower/arduino-littlefs-upload/blob/24e2afdc60c22614c0261cf2a5f15809876dfd7b/src/extension.ts#L257-L261

earlephilhower, 2024-11-02:
And your example is even worse, it's skipping offsets.
> Partitions with blank offsets in the CSV file will start after the previous partition, or after the partition table in the case of the first partition.

So you need an actual stateful FSM to parse and process the CSV.  Joy! :sob:

commit 783f98ee (2024-11-02): Redo ESP32 partitions.csv parsing

Support comments ('#'), -K and -M modifiers, don't skip 1st line,
and match on the "type" and note "name" fields.   Match "LittleFS"
type, too.

Fixes #61
commit 347d5c52 (2024-11-02): Redo ESP32 partitions.csv parsing

Support comments ('#'), -K and -M modifiers, don't skip 1st line,
and match on the "type" and note "name" fields.   Match "LittleFS"
type, too.

Fixes #61
PR #62 (2024-11-03): Redo ESP32 partitions.csv parsing
Support comments ('#'), -K and -M modifiers, don't skip 1st line,
and match on the "type" and note "name" fields.   Match "LittleFS"
type, too.

Fixes #61
commit 71a20b86 (2024-11-03): Redo ESP32 partitions.csv parsing (#62)

Support comments ('#'), -K and -M modifiers, don't skip 1st line,
and match on the "type" and note "name" fields.   Match "LittleFS"
type, too.

Fixes #61
closed 2024-11-03 by earlephilhower