earlephilhower/arduino-littlefs-upload #64: Having issues accessing files via web site

godmcse, 2024-11-03:
I have spent the last 2 days working on this.  I came across arduino-littlefs-upload and the files seem to upload.  however I can not access them from the web page.  I uploaded a config file,

I test using this code and the file is found:
` File file = LittleFS.open("/config.json", FILE_WRITE);`

however, the webpage cannot find the file.  I uploaded a favicon.gif file and that too is not found.

I tried ./config.json and ./data/config.json I keep getting errors.  what am I doing wrong.  

The plugin does show the files are uploaded successfully.

Any help would be great.

earlephilhower, 2024-11-03:
Please make a small MCVE that shows the issue, or else it's unlikely that anybody can replicate your problem or help you.  We'll need the partitions.csv and any other scripts or files to build.

In any case, as someone who doesn't use the ESP32, this doesn't look like a filesystem upload issue at all.  IIRC, HTTP status `500` is a *server error*.  You'd just get a `404 Not Found` if the web server was working properly but just didn't have that file.

godmcse, 2024-11-03:
> Please make a small MCVE that shows the issue, or else it's unlikely that anybody can replicate your problem or help you. We'll need the partitions.csv and any other scripts or files to build.
>
> In [...]

i hope this helps
I am trying to control a step motor and i want to access the Config file.  but the file is not accessible.  

```
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <Stepper.h>
#include <LittleFS.h>
//#include <ArduinoJson.h>

//config file
// const String config_filname ="/config.json";

int Bun0 = 0;
int BunUser = 500;
int Arch0 = 0;
int ArchUser = 500;
int Onerev = 360;
String user_string = "";

// Stepper Motor Settings
const int stepsPerRevolution = 2048;  // change this to fit the number of steps per revolution
#define IN1 19
#define IN2 18
#define IN3 5
#define IN4 17
Stepper myStepper(stepsPerRevolution, IN1, IN3, IN2, IN4);

// Replace with your network credentials
const char* ssid = "Dynimage";
const char* password = "JLJA-2002";

const char* ssidAP = "ESP32-Access-Point";
const char* passwordAP = "1234";

IPAddress staticIP(192, 168, 10, 45);     // ESP32 static IP
IPAddress gateway(192, 168, 10, 1);       // IP Address of your network gateway (router)
IPAddress subnet(255, 255, 255, 0);       // Subnet mask
IPAddress primaryDNS(192, 168, 10, 1);    // Primary DNS (optional)
IPAddress secondaryDNS(192, 168, 10, 3);  // Secondary DNS (optional)

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

// Search for parameters in HTTP POST request
const char* PARAM_INPUT_1 = "direction";
const char* PARAM_INPUT_2 = "steps";

// Variables to save values from HTML form
String direction;
String steps;

// Variable to detect whether a new request occurred
bool newRequest = false;

// HTML to build the web page
const char index_html[] PROGMEM = R"rawliteral(
<!DOCTYPE html>
<html>
<head>
  <title>Shoe Control test</title>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="shortcut icon" type="image/gif" href="./data/favicon.gif"> 
</head>
[... 142 lines cut ...]
        }
      }
    }
    request->send(200, "text/html", index_html);
    newRequest = true;
  });

  server.begin();
}

void loop() {
  // Check if there was a new request and move the stepper accordingly
  if (newRequest) {
    if (direction == "CW") {
      // Spin the stepper clockwise direction
      myStepper.step(steps.toInt());
    } else {
      // Spin the stepper counterclockwise direction
      myStepper.step(-steps.toInt());
    }
    newRequest = false;
  }
}

```

earlephilhower, 2024-11-03:
Okay, that's actually pretty large and using lots of other libraries that aren't easily used outside of your configl

Can you write a simple ESP32 app that opens up and reads out `config.json` from LittleFS to the Serial port on startup?  That works, then your issue is best opened on the ESP32 Arduino core.  That fails, we can repro here and see what's up.

godmcse, 2024-11-03:
I even have do a simple codes like this to test yet I can not find the files.   the files list shows the files are on the device, but I can not access them via the URL.  

`
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>

const String default_ssid = "WIFI";
const String default_wifipassword = "PASSWORD";
const int default_webserverporthttp = 80;

const char index_html[] PROGMEM = R"rawliteral(
<!DOCTYPE HTML>
<html lang="en">
<head>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <meta charset="UTF-8">
</head>
<body>
  <p><h1>File Upload</h1></p>
  <p>Free Storage: %FREESPIFFS% | Used Storage: %USEDSPIFFS% | Total Storage: %TOTALSPIFFS%</p>
  <form method="POST" action="/upload" enctype="multipart/form-data"><input type="file" name="data"/><input type="submit" name="upload" value="Upload" title="Upload File"></form>
  <p>After clicking upload it will take some time for the file to firstly upload and then be written to SPIFFS, there is no indicator that the upload began.  Please be patient.</p>
  <p>Once uploaded the page will refresh and the newly uploaded file will appear in the file list.</p>
  <p>If a file does not appear, it will be because the file was too big, or had unusual characters in the file name (like spaces).</p>
  <p>You can see the progress of the upload by watching the serial output.</p>
  <p>%FILELIST%</p>
</body>
</html>
)rawliteral";

String listFiles(bool ishtml = false);

// configuration structure
struct Config {
  String ssid;               // wifi ssid
  String wifipassword;       // wifi password
  int webserverporthttp;     // http port number for web admin
};

// variables
Config config;                        // configuration
AsyncWebServer *server;               // initialise webserver

void setup() {
  Serial.begin(115200);

  Serial.println("Booting ...");

  Serial.println("Mounting SPIFFS ...");
  if (!SPIFFS.begin(true)) {
    // if you have not used SPIFFS before on a ESP32, it will show this error.
    // after a reboot SPIFFS will be configured and will happily work.
    Serial.println("ERROR: Cannot mount SPIFFS, Rebooting");
    rebootESP("ERROR: Cannot mount SPIFFS, Rebooting");
  }

  Serial.print("SPIFFS Free: "); Serial.println(humanReadableSize((SPIFFS.totalBytes() - SPIFFS.usedBytes())));
  Serial.print("SPIFFS Used: "); Serial.println(humanReadableSize(SPIFFS.usedBytes()));
  Serial.print("SPIFFS Total: "); Serial.println(humanReadableSize(SPIFFS.totalBytes()));

  Serial.println(listFiles());

  Serial.println("\nLoading Configuration ...");

  config.ssid = default_ssid;
  config.wifipassword = default_wifipassword;
  config.webserverporthttp = default_webserverporthttp;

[... 106 lines cut ...]
    // close the file handle as the upload is now done
    request->_tempFile.close();
    Serial.println(logmessage);
    request->redirect("/");
  }
}

String processor(const String& var) {
  if (var == "FILELIST") {
    return listFiles(true);
  }
  if (var == "FREESPIFFS") {
    return humanReadableSize((SPIFFS.totalBytes() - SPIFFS.usedBytes()));
  }

  if (var == "USEDSPIFFS") {
    return humanReadableSize(SPIFFS.usedBytes());
  }

  if (var == "TOTALSPIFFS") {
    return humanReadableSize(SPIFFS.totalBytes());
  }

  return String();
}`

earlephilhower, 2024-11-03:
I suggest for this you use the [ESP32 Arduino repo](https://github.com/espressif/arduino-esp32), since you're sure the files are up and running.  There's lots more folks (and the authors) who can help you...

godmcse, 2024-11-03:
thank you will try it.,

godmcse, 2024-11-04:
fyi.  i FOUND A SAMPLE THAT WORKS AND WORKING WITH THAT.  THANKS AGAIN FOR THE HELP.

closed 2024-11-04 by godmcse