earlephilhower/arduino-littlefs-upload #67: Baud rate prolem

F4HSL-Yves, 2024-11-22:
Hello, i have installed the plugin on arduino ide 2.3 (windows11). When i use the command “upload little fs” i get an error about the baud rate option :

esptool: error: argument –baud/-b: invalid arg_auto_int value: ‘NaN’
ERROR: Upload failed, error code: 2

Any idea ?
Best Regards
Yves

earlephilhower, 2024-11-22:
I think your board only does DFU and not esptool/serial.  That's not compatible with the uploader.  There's no workaround other using a more standard board.

closed 2024-11-22 by earlephilhower

F4HSL-Yves, 2024-11-22:
Ok Tnx, my board is the standard arduino nano esp32, anyway which similar board is compatible with your plugin ?

F4HSL-Yves, 2024-11-22:
@per1234, tnx a lot, i will try your solution.

earlephilhower, 2024-11-22:
I've also added a link to that comment in the README.md file.  Thanks, @per1234 !

On a side note, it's not that hard to add add'l upload methods, but I don't have anything that uses DFU upload. I assume that mode can place arbitrary data in arbitrary spots, so if someone has a command line sample for uploading a FS then it'd be possible to add it in (like the OTA and picotool methods just done).  I'd also assume there's a `board.xxxx` variable I can reference to know it's DFU and not normal esptool...