earlephilhower/arduino-littlefs-upload #69: ERROR: Partition file not found!

mkinney88, 2024-12-20:
When using the SparkFun ESP32 Things Plus it I get this error when I select the large_spiffs partition. Also, when I use the default selection for 16mb partition it will flash the default esp32 partition and not the default_16MB partition scheme. To replicate it just select the SparkFun ESP32 Things Plus under the board in the IDE and try and flash the littlefs, you don't need a board connected to see it is using the wrong partition. I checked the boards.txt file in the C:\Users\matt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\boards.txt and the board does reference the correct partition, but it seems that this extension isn't.  

LittleFS Filesystem Uploader v1.5.0 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: C:\Users\matt\Documents\Arduino\ESP32_Now_AP_Test_V8
   Data Path: C:\Users\matt\Documents\Arduino\ESP32_Now_AP_Test_V8\data
      Device: ESP32 series, model esp32
Using partition: large_spiffs
  Partitions: C:\Users\matt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\tools\partitions\large_spiffs.csv

ERROR: Partition file not found!

earlephilhower, 2024-12-20:
Try 1.5.1.  It now selects default_16MB or large_spiffs_16MB for your board.

closed 2024-12-20 by earlephilhower

mkinney88, 2024-12-20:
thank you! works perfectly now.

commit ade38745 (2024-12-20): Double-indirect the ESP32 partition file name

Fixes #69

The partition name menu selection's value only points to the proper line of
build.properties, and is not always the file name for ESP32 partitions.
Use the label to indirect to the board props to select the right file.
PR #71 (2024-12-20): Double-indirect the ESP32 partition file name
Fixes #69

The partition name menu selection's value only points to the proper line of build.properties, and is not always the file name for ESP32 partitions. Use the label to indirect to the board props to select the right file.
commit ed3bf53c (2024-12-20): Double-indirect the ESP32 partition file name (#71)

Fixes #69

The partition name menu selection's value only points to the proper line of
build.properties, and is not always the file name for ESP32 partitions.
Use the label to indirect to the board props to select the right file.