earlephilhower/arduino-littlefs-upload #70: LittleFS Filesystem Uploader v1.5.0 keeps failing

pdfelen, 2024-12-20:
Thanks for spending effort fixing #17 
Today I updated my IDE, board packs and this tool to try out the solution, but did not manage to get it working. The normal OTA update is working, but LittleFS keeps getting the same error. As written in the Glitch section, I have tried selecting a different board or opening it as a second sketch. Unfortunately for me the result remains the same.

Any additional tips or solutions?

**Hereby the Arduino IDE versions:**
Version: 2.3.4
Date: 2024-12-03T10:03:03.490Z
CLI Version: 1.1.1

**Hereby the tool output:**
LittleFS Filesystem Uploader v1.5.0 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: <project-path>\MyProject
   Data Path: <project-path>\MyProject\data
      Device: ESP8266 series
Network Info: <ip-address>:8266

Building LittleFS filesystem
Command Line: <tool-parth>\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933\mklittlefs.exe -c <project-path>\MyProject\data -p 256 -b 8192 -s 2072576 <temp-path>\tmp-11120-4AWVcDauxaqA-.littlefs.bin
/file1.css
/file2.svg
/file3.gif
/file4.gif
/file5.htm
/file6.gif
/file7.ico
/file8.htm
/file9.gif
/file10.htm
/file11.htm

Uploading LittleFS filesystem
Command Line: <tool-path>\Arduino15\packages\esp8266\tools\python3\3.7.2-post1\python3.exe <tool-path>\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\tools\upload.py --chip esp8266 --port 192.168.1.118 --baud 115200 write_flash 2097152 <temp-path>\tmp-11120-4AWVcDauxaqA-.littlefs.bin
esptool.py v3.0
Serial port 192.168.1.118

A fatal esptool.py error occurred: could not open port '192.168.1.118': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
Completed upload.

earlephilhower, 2024-12-20:
OTA on the ESP8266 is not supported.  It's there for the Pico only.  If you'd like to make a PR adding it, that would be great, but the 8266 is basically EOL at this point so not something I'm actively working on.

closed 2024-12-20 by earlephilhower

pdfelen, 2024-12-20:
Hi Earlephilpower,
I appreciate your work and dedication.

In #17 I requested support for ESP8266 in the new API, this was added. I understand only for serial.
Before it was also working for OTA. I'm willing to spend some time on it in the coming weeks. How can I support?

earlephilhower, 2024-12-20:
If you fork this repo, you can build the extension just using GitHub CI, nothing needs to even run on your machine to make a VSIX for testing.  So, development is pretty simple.  Just edit the `src/extension.ts` and commit to a branch and run using GH actions.

For OTA on the 8266 you probably just need to update the command line.  See the Pico implementation
https://github.com/earlephilhower/arduino-littlefs-upload/blob/ed3bf53c7442aa63f01e9b6c4e917df57fc404de/src/extension.ts#L481-L487.  I'd check for `(network)` and then update the command line as required depending on local or network port.  It may be almost a simple cut-n-paste, even, since I did the 8266 core waaaaaay back when and the Pico core today...

pdfelen, 2024-12-21:
Hi Earle,
Thanks for your tips. I finished it and tested it and released it as v1.5.1.b01 in my fork. It works fine. Only minor detail is that the update dots are only visible after processing. But I think this has more to do with something else.
Would be nice to get it into your main line. Feel free to backport.
Kind regards,
Pepijn

earlephilhower, 2024-12-21:
Nice!  You can make a PR here, just hit the option from the main repo page, https://github.com/earlephilhower/arduino-littlefs-upload/.  Should be an option on the top of the page since you have a fork+branch with changes.

The update dots are related to the IDE.  It's buffering stdout, so there's not much we can do here...

pdfelen, 2024-12-21:
With the dots: as expected.
PR Done as you can see above.