earlephilhower/arduino-littlefs-upload #72: does mklittlefs ignore timestamps?

georgepapas123, 2024-12-20:
When creating the image to flash, mklittlefs seems to ignore timestamps. 
(all files have 1970-01-01 00:00:00 timestamp)
If a file is later changed and saved, then the timestamp is correct.
Is this a bug or am I doing something wrong?

I use the 'standard' function to check timestamps.

int listDir(const char *dirname) {
  int filecount = 0;
  Serial.printf("Listing directory: %s\n", dirname);
  Dir root = LittleFS.openDir(dirname);
  while (root.next()) {
    filecount++;
    File file = root.openFile("r");
    Serial.print("  FILE: ");
    Serial.print(root.fileName());
    Serial.print("  SIZE: ");
    Serial.print(file.size());
    time_t cr = file.getCreationTime();  // 0 = 1970-01-01 00:00:00
    time_t lw = file.getLastWrite();     
    file.close();
    struct tm *tmstruct = localtime(&cr);
    Serial.printf("    CREATION: %d-%02d-%02d %02d:%02d:%02d\n", (tmstruct->tm_year) + 1900, (tmstruct->tm_mon) + 1, tmstruct->tm_mday, tmstruct->tm_hour, tmstruct->tm_min, tmstruct->tm_sec);
    tmstruct = localtime(&lw);
    Serial.printf("  LAST WRITE: %d-%02d-%02d %02d:%02d:%02d\n", (tmstruct->tm_year) + 1900, (tmstruct->tm_mon) + 1, tmstruct->tm_mday, tmstruct->tm_hour, tmstruct->tm_min, tmstruct->tm_sec);
  }
  return filecount;
}

earlephilhower, 2024-12-20:
The ESP8266 mklittlefs might not include or support that feature, but the Pico does.  It's done automatically by mklittlefs and the plugin here doesn't have any influence.  

https://github.com/earlephilhower/mklittlefs/commit/a5709145b60e3f5ab2eb679c468229e2ac77e525

You should check the version of mklittlefs being run.  It shows the whole command line in the IDE so you can run it from a CMD prompt with `--version` .  If it's not from esp-quick-toolchain (i.e. you're running the ESP32 branch) I can't point you at anything, but on the ESP8266 one the above patch was there (patch in 2020, my last rebuild of the toolchain was 2023).

closed 2024-12-20 by earlephilhower

earlephilhower, 2024-12-20:
One word of warning, BTW.  The latest revs of mklittlefs are only compatible with the Pico and ESP32.  LittleFS was updated and the maximum file name length was reverted to the "standard" 256 bytes.  The ESP8266 only runs with namelen of 32 without patching (which eats RAM).

georgepapas123, 2024-12-20:
Thank you for the immediate reply!
here's what I have:
\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec\mklittlefs --version
mklittlefs  version: 0.2.3-48-g30b7fc1

\Arduino15\packages\esp8266\tools\mklittlefs\3.1.0-gcc10.3-e5f9fec\package.json 
{
   "description": "littlefs-utility",
   "name": "mklittlefs",
   "system": [ "windows_amd64" ],
   "url": "https://github.com/earlephilhower/esp-quick-toolchain",
   "version": "5.100300.220621"
}

earlephilhower, 2024-12-21:
Hmmm, that version should add the attributes (no native FS timestamp so I choose the 'c' and 't' attributes).  The ESP8266 Arduino code does check for them (since I did both changes ~3 years ago) in the littlefs lib:
https://github.com/esp8266/Arduino/blob/bb79e9076e8402c81c24efbabd19f1baab1c6bd1/libraries/LittleFS/src/LittleFS.h#L525-L533

So I did some digging this morning and it's a case of "really old source" running on the host(!!!) and "cutting edge software" on the embedded system.  :laughing: 

`time_t` went from 32->64 bits to avoid the year 2038 problem.  I moved Newlib up to using it on both the Pico and ESP8266 way back when.

`mklittlefs` is built to force time_t to 32-bits because when it was first done the PC had 64-bit time, but Newlib did not.
https://github.com/earlephilhower/mklittlefs/blob/c15c22a18ea35f1e5f12d53d6f4c968dc9b36460/main.cpp#L167

So this is a problem in mklittlefs, not here, but there is a simple SW solution on the ESP or Pico.  Right now it's checking that the 'c' time attribute == sizeof(time_t) == 8, but it's getting 4 and seeing that as an error.  Instead, the LittleFS.h header can check if it gets a 4-byte time and read it into a 32b and assign to the real 64b time.

For example:
````
-Subproject commit bf7742310b422977d245833815fd46275cefdebe
+Subproject commit bf7742310b422977d245833815fd46275cefdebe-dirty
diff --git a/libraries/LittleFS/src/LittleFS.h b/libraries/LittleFS/src/LittleFS.h
index 162edf1f1..8f515ba55 100644
--- a/libraries/LittleFS/src/LittleFS.h
+++ b/libraries/LittleFS/src/LittleFS.h
@@ -498,7 +498,12 @@ public:
         time_t ftime = 0;
         if (_opened && _fd) {
             int rc = lfs_getattr(_fs->getFS(), _name.get(), 'c', (void *)&ftime, sizeof(ftime));
-            if (rc != sizeof(ftime)) {
+            if (rc == sizeof(int32_t)) {
+                int32_t time32;
+                lfs_getattr(_fs->getFS(), _name.get(), 'c', (void *)&time32, sizeof(time32));
+                ftime = time32;
+            } else {
+                Serial.printf("rc=%d\n", rc);
                 ftime = 0;    // Error, so clear read value
             }
         }
````

You need to apply that to the 't' and the root directory creationtime call for sanity's sake, but it's relatively straightforward.

So this is not a plugin issue but a MklittleFS one.  I'll open an issue and fix there referencing this for posterity.

Thx for finding this!  I'm really surprised nobody's noticed, or at least been curious enough to look into it...

earlephilhower, 2024-12-21:
And just to make it more interesting, the timestamp on the "/" root directory was actually correct (by accident) as it always was sizeof(time_t)==8.  Anyway, the SW change to LittleFS above can get you going on the 8266 since you can't use any updated mklittlefs versions there.

georgepapas123, 2024-12-21:
Thanks a lot for you time and effort for getting to the bottom of this.

georgepapas123, 2024-12-21:
The mod to LittleFS.h almost worked.
All creation timestamps were fixed (for files in the original image created my mklittlefs).
But modification timestamps (for files modified afterwards) were now 0.
The following adjustment was required:
```C++
            if (rc != sizeof(ftime)) {
              if (rc == sizeof(int32_t)) {
                   int32_t time32;
                   lfs_getattr(_fs->getFS(), _name.get(), 'c', (void *)&time32, sizeof(time32));
                   ftime = time32;
               } else {
                   Serial.printf("rc=%d\n", rc);
                   ftime = 0;    // Error, so clear read value
               }
            }
        }
```
(similarly for 't')
Now both mklittlefs original timestamps and those set by LittleFS library code later are shown correctly.
Thanks again for the fix.