earlephilhower/arduino-littlefs-upload #76: esptool - No such file or directory

icho40, 2025-01-01:
Hi,
I have the same problem as [aziemer.](https://github.com/earlephilhower/arduino-littlefs-upload/issues/23#issue-2265670663)
esptool is installed and working.

pip show esptool:
Name: esptool
Version: 4.8.1

When I try to upload something to the esp, I get this error:

> LittleFS Filesystem Uploader v1.5.1 -- https://github.com/earlephilhower/arduino-littlefs-upload
> 
> Uploading LittleFS filesystem
> Command Line: python3 /home/willi/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 /tmp/tmp-1901608-03GfhMX7zNU1-.littlefs.bin
> python3: can't open file '/home/willi/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py': [Errno 2] No such file or directory
> 
> ERROR:  Upload failed, error code: 2

When I create a symbolic link from esptool.py to esptool,

> 15:04:27 willi@icho:4.9.dev3 ls -l esptool.py 
> lrwxrwxrwx 1 willi willi 7  1. Jan 15:00 esptool.py -> esptool

 I get this error:

> LittleFS Filesystem Uploader v1.5.1 -- https://github.com/earlephilhower/arduino-littlefs-upload
> 
> Uploading LittleFS filesystem
> Command Line: python3 /home/willi/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 /tmp/tmp-1901608-FtiedAzwNetk-.littlefs.bin
> SyntaxError: Non-UTF-8 code starting with '\xba' in file /home/willi/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py on line 1, but no encoding declared; see https://peps.python.org/pep-0263/ for details
> 
> ERROR:  Upload failed, error code: 1

Arduino-AppImage:
Version: 2.3.3
Date: 2024-09-25T09:35:36.309Z
CLI Version: 1.0.4

OS:
MX-Linux
Version:24.12.04

earlephilhower, 2025-01-01:
You can't make a symlink to `esptool` because it's evidently an executable image and we explicitly call the python interpreter under Linux.

In fact, we must never use a ` pip3` installed version of esptool, only the one shipped with the ESP32 Arduino core tools.  We actually query the Arduino core for the location of the tool which should be installed by it as part of the toolchain.

The 4.6 release of the tool does have the esptool.py installed on my system
````
$ ls ~/.arduino15/packages/esp32/tools/esptool_py/4.6/
esptool  esptool.py
````

However once I updated to the 3.1 Arduino core and `4.9dev3` I see they are now shipping **binary compiled versions** of `esptool`.

I have a lot of respect for the ESP32 Arduino guys' work, but damn they like to change stuff just for the hell of it, making it a real pain for developers.  See also the not-quite-compatible I2S driver updates and the WiFiClient->NetworkClient renaming (I get why on this one, but a `using` would've been nice to not break compatibility with every other Arduino core).  :sob:

So the normal way of calling `esptool` for earlier versions won't work on 3.1.  Yet more special casing will be required...

icho40, 2025-01-01:
Thanks for the explanation.
I have now got it working, so this change no longer bothers me.

Thanks for your work.

closed 2025-01-01 by earlephilhower

icho40, 2025-01-01:
Now I have another problem:

```
LittleFS Filesystem Uploader v1.5.3 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: /home/willi/Arduino/ESP32_Stepper_Motor_Websocket
   Data Path: /home/willi/Arduino/ESP32_Stepper_Motor_Websocket/data
      Device: ESP32 series, model esp32
Using partition: default
  Partitions: /home/willi/.arduino15/packages/esp32/hardware/esp32/3.1.0/tools/partitions/default.csv
       Start: 0x290000
         End: 0x3f0000
 Serial Port: /dev/ttyUSB0

Building LittleFS filesystem
Command Line: /home/willi/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933/mklittlefs -c /home/willi/Arduino/ESP32_Stepper_Motor_Websocket/data -p 256 -b 4096 -s 1441792 /tmp/tmp-1960356-2q6AbNAYFt2H-.littlefs.bin
/index.html
/style.css
/script.js
/favicon.png

Uploading LittleFS filesystem
Command Line: python3 /home/willi/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 /tmp/tmp-1960356-2q6AbNAYFt2H-.littlefs.bin
SyntaxError: Non-UTF-8 code starting with '\xba' in file /home/willi/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py on line 1, but no encoding declared; see https://peps.python.org/pep-0263/ for details

ERROR:  Upload failed, error code: 1
```

earlephilhower, 2025-01-01:
Remove the symlink you manually made in that directory.  There is no real `.../4.9.dev3/esptool.py` shipped, that only exists in earlier ones.

PR #77 (2025-01-01): ESP32 Arduino now ships binary esptool under Linux
Sometimes we'll have a PY, sometimes we'll have an ELF.  Use hueristics to pick the right version at runtime.

Fixes #76
commit 4c74c4c3 (2025-01-01): ESP32 Arduino now ships binary esptool under Linux (#77)

Sometimes we'll have a PY, sometimes we'll have an ELF.  Use hueristics
to pick the right version at runtime.

Fixes #76