earlephilhower/arduino-littlefs-upload #78: A fatal error occurred: Invalid head of packet (0x70): Possible serial noise or corruption.

CrossScarDev, 2025-01-04:
I'm using an Adafruit Feather ESP32 v2.

Log:
```
Uploading LittleFS filesystem
Command Line: python3 /home/crossscar/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool.py --chip esp32 --port /dev/ttyACM0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 6750208 /tmp/tmp-8780-NAeO4dbMftdC-.littlefs.bin
esptool.py v4.6
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-PICO-V3-02 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, Embedded PSRAM, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 14:2b:2f:af:67:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00670000 to 0x007effff...
Compressed 1572864 bytes to 2486...
Writing at 0x00670000... (100 %)

A fatal error occurred: Invalid head of packet (0x70): Possible serial noise or corruption.

ERROR:  Upload failed, error code: 2
```

earlephilhower, 2025-01-04:
That error's generated by esptool, not the plug in.  I'd try changing baud rates (lower), double-checking the flash modes DIO/QIO, and maybe seeing if smaller filesystems can be uploaded.  The ESP32 Arduino core might be able to get you more help.  Good luck!

closed 2025-01-04 by earlephilhower

CrossScarDev, 2025-01-04:
I just retried today and it randomly worked.

CrossScarDev, 2025-03-05:
It now seems to work fine with a normal fast flash speed.