earlephilhower/arduino-littlefs-upload #82: Esp32 HUGE APP error

HamzaYslmn, 2025-02-08:
[image]

Esp32 Config:

[image]

TEST CODE:

[image]

TREE:

[image]

OUTPUT:

18:27:30.248 -> Failed to open file for reading

HamzaYslmn, 2025-02-08:
https://randomnerdtutorials.com/arduino-ide-2-install-esp32-littlefs/

ı use this tutorial

earlephilhower, 2025-02-08:
I have no idea what you're running, but you're not using this plug in.  Your first image (next time, please just paste text and not pictures of text) shows **mkspiffs**.  So, you're building a SPIFFS FS, not a LittleFS one.  So your app's not going to be able to read it...

closed 2025-02-08 by earlephilhower

HamzaYslmn, 2025-02-08:
SPIFFS Filesystem Uploader

Using partition: default
Building SPIFFS filesystem
C:\Users\ASUS\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3/mkspiffs.exe -c C:\Users\ASUS\Documents\VsCode\AtikSu\Mountain-Atik-SU\test\FS/data -p 256 -b 4096 -s 1441792 C:\Users\ASUS\AppData\Local\Temp\tmp-20108-YNBJZzbMXAKD-.spiffs.bin
/fs.txt

Uploading SPIFFS filesystem
C:\Users\ASUS\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3/esptool.exe --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 C:\Users\ASUS\AppData\Local\Temp\tmp-20108-YNBJZzbMXAKD-.spiffs.bin
esptool.py v4.8.1
Serial port COM5
Connecting.......
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c0:49:ef:d4:54:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00290000 to 0x003effff...
Compressed 1441792 bytes to 2727...
Writing at 0x00290000... (100 %)
Wrote 1441792 bytes (2727 compressed) at 0x00290000 in 8.1 seconds (effective 1431.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Completed upload.

HamzaYslmn, 2025-02-08:
My Code for littlefs:

```
#include "LittleFS.h"
 
void setup() {
  Serial.begin(115200);
  
  if(!LittleFS.begin()){
    Serial.println("An Error has occurred while mounting LittleFS");
    return;
  }
  
  File file = LittleFS.open("/fs.txt", "r");
  if(!file){
    Serial.println("Failed to open file for reading");
    return;
  }
  
  Serial.println("File Content:");
  while(file.available()){
    Serial.write(file.read());
  }
  file.close();
}
 
void loop() {

}
```

HamzaYslmn, 2025-02-08:
> but you're not using this plug in

I apologize for not using your latest version, it's solved now.