earlephilhower/arduino-littlefs-upload #84: OTA upload not working on ESP32

MrMabulous, 2025-03-01:
I'm trying to upload LittleFS using OTA.

ArduinoOTA code is added to the project and ota upload of sketches works without problem. LittleFS upload through USB also works without problem. However, if I try LittleFS upload via OTA, the LittleFS upload hangs on the following prompt:

`LittleFS Filesystem Uploader v1.5.3 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: C:\Users\Username\Desktop\Esp32OledTest\OledTest
   Data Path: C:\Users\Username\Desktop\Esp32OledTest\OledTest\data
      Device: ESP32 series, model esp32c3
Using partition: min_spiffs
  Partitions: C:\Users\Username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\tools\partitions\min_spiffs.csv
       Start: 0x3d0000
         End: 0x3f0000
Network Info: 192.168.1.78:3232

Building LittleFS filesystem
Command Line: C:\Users\Username\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933\mklittlefs.exe -c C:\Users\Username\Desktop\Esp32OledTest\OledTest\data -p 256 -b 4096 -s 131072 C:\Users\Username\AppData\Local\Temp\tmp-2544-9V57MZRqHNvc-.littlefs.bin
/configuration.html
/config_script.js
/config_styles.css
/index.html
/index_styles.css
/script.js

Uploading LittleFS filesystem
Command Line: C:\Users\Username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\tools\espota.py -r -i 192.168.1.78 -p 3232 -f C:\Users\Username\AppData\Local\Temp\tmp-2544-9V57MZRqHNvc-.littlefs.bin -s`

I can see that espota.py is called without password argument, so I removed the ArduinoOTA.setPassword(...) line from the code but the LittleFS upload still hangs on the espota.py call

on the ESP32 side ArduinoOTA.onStart() is never called

I'm using an ESP32-C3

MrMabulous, 2025-03-01:
After changing "espota.py" to "espota.exe" and executing the command `C:\Users\Username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\tools\espota.exe -r -i 192.168.1.78 -p 3232 -f C:\Users\Username\AppData\Local\Temp\tmp-2544-9V57MZRqHNvc-.littlefs.bin -s` from the commandline the upload worked. Does espota.py need to be replaced with espota.exe in src/extension.ts on windows platforms maybe?

earlephilhower, 2025-03-02:
Nice debug!

The ESP folks change between Python and compiled Python pretty frequently so we had the same issue pop up with the latest version of the core with simple `esptool`.  But here, I think on Windows they always compiled things so this was always an issue for OTA.

In fact, they ship a precompiled Windows executable in the core itself alongside the Python version they use everywhere else.  That's just funky IMHO...exes were always shipped as part of the hardware tools binaries, not in the source code.  (And 10K for the Python source vs **5.9MB** for the Windows exe....ouch!)

````
ls -l ~/.arduino15/packages/esp32/hardware/esp32/3.1.3/tools/
total 17956
-rw-r--r-- 1 earle earle 5923784 Feb 22 09:46 espota.exe
-rwxr-xr-x 1 earle earle   10424 Feb 22 09:46 espota.py
-rw-r--r-- 1 earle earle 6389808 Feb 22 09:46 gen_esp32part.exe
-rwxr-xr-x 1 earle earle   23891 Feb 22 09:46 gen_esp32part.py
-rwxr-xr-x 1 earle earle 6006318 Feb 22 09:46 gen_insights_package.exe
-rw-r--r-- 1 earle earle    2207 Feb 22 09:46 gen_insights_package.py
drwxr-xr-x 3 earle earle    4096 Feb 22 09:46 ide-debug
drwxr-xr-x 2 earle earle    4096 Feb 22 09:46 partitions
-rw-r--r-- 1 earle earle    8078 Feb 22 09:46 pioarduino-build.py
````

In any case, since you found the issue and a solution would you like to make a PR here?  If not, I can implement it easily enough.  Only gotcha is you need to check the OS and either run `python3 espota.py` or `expota.exe`...

MrMabulous, 2025-03-04:
Implemented here, but I'm not sure whether it might break macOS https://github.com/earlephilhower/arduino-littlefs-upload/pull/85

closed 2025-03-04 by earlephilhower