earlephilhower/arduino-littlefs-upload #90: The LittleFS uploader seems to ignore the partition size

MikeyMoMo, 2025-03-20:
I have a 2MB (and growing) upload that works just fine on IDE 1.x but fails on IDE 2.x  (currently on v2.3.4) because it says the partition is too small.  The partition is 5 times the size of the upload.  This tells me that the partition size is assumed to be something smaller than 2MB.  Possibly it is using 1.5MB.  Don't know.

Is there a way around this problem or must the developer fix the code to honor the size of the partition and the partition.csv file that is in the sketch directory?

I will have to continue using IDE 1.x until this has been resolved.  Please attend to it soon.

Here is my partition table:

```
# Name,   Type, SubType,  Offset,  Size,    Flags
nvs,      data, nvs,      36K,     20K,
otadata,  data, ota,      56K,     8K,
app0,     app,  ota_0,    64K,     3M,
# The theory is that this will work.
#app1,     app,  ota_1,    ,        3M,
#spiffs,   data, spiffs,   ,        6M,
spiffs,   data, spiffs,   ,        10M,
coredump, data, coredump, ,        0x10000,
```

And here is what the program shows, waiting for files that will never arrive...

```
LittleFS Info:
LittleFS Total Bytes: 10485760
LittleFS Used Bytes: 8192
LittleFS Free Bytes: 10477568
```
The 8192 seems to be overhead in the otherwise empty partition.  This is just after the area has been formatted.

And here is the output of the load attempt:
```
LittleFS Filesystem Uploader v1.5.3 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: D:\Arduino\@ESP32 Sketches\TTGO T-Display S3\Clock & Date with BG Pix and Dimming\T_Display_S3_BG_Clock_v1.02
   Data Path: D:\Arduino\@ESP32 Sketches\TTGO T-Display S3\Clock & Date with BG Pix and Dimming\T_Display_S3_BG_Clock_v1.02\data
      Device: ESP32 series, model esp32s3
Using partition: partitions.csv in sketch folder
  Partitions: D:\Arduino\@ESP32 Sketches\TTGO T-Display S3\Clock & Date with BG Pix and Dimming\T_Display_S3_BG_Clock_v1.02\partitions.csv
       Start: 0xc10000
         End: 0x1610000
 Serial Port: COM23

Building LittleFS filesystem
Command Line: C:\Users\bangu\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933\mklittlefs.exe -c D:\Arduino\@ESP32 Sketches\TTGO T-Display S3\Clock & Date with BG Pix and Dimming\T_Display_S3_BG_Clock_v1.02\data -p 256 -b 4096 -s 10485760 C:\Users\bangu\AppData\Local\Temp\tmp-10308-Pf030tS7y2dj-.littlefs.bin
/Arches.jpg
/BigRack.jpg
...and more...
/YellowOrangeSwirl.jpg
/YellowWash.jpg

Uploading LittleFS filesystem
Command Line: C:\Users\bangu\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.9.dev3\esptool.exe --chip esp32s3 --port COM23 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 12648448 C:\Users\bangu\AppData\Local\Temp\tmp-10308-Pf030tS7y2dj-.littlefs.bin
esptool.py v4.8.1
Serial port COM23
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: e4:b0:63:8a:ca:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB

A fatal error occurred: File C:\Users\bangu\AppData\Local\Temp\tmp-10308-Pf030tS7y2dj-.littlefs.bin (length 10485760) at offset 12648448 will not fit in 16777216 bytes of flash. Change the --flash_size argument, or flashing address.

ERROR:  Upload failed, error code: 2
```
It claims to see the partitions.csv file but then seems to ignore the setting.  The total of the load is about 2MB and loads just fine in the IDE 1.x program.

earlephilhower, 2025-03-20:
The plugin does parse the requested partition (it's 10M in your example, not 2M), but it unfortunately isn't handling your commented lines well due to a bug.  So it parses 3M app0,  3M app1, 6M spiffs(1), and the final, real, 10M spiffs(2).  That spiffs(2) will begin at 3M+3M+6M=10M = 0xc10000 as shown, which will mean it tries to write from 10M..20M on the upload.

closed 2025-03-20 by earlephilhower

MikeyMoMo, 2025-03-20:
> completed via #91
> <earlephilhower/arduino-littlefs-upload/issue/90/issue_event/16899554950@
> github.com>

Removing the comments fixed it. Please update the code to honor the format
of the file (ignore comment lines).   Thanks for the help and the speedy
answer to get this working.

PR #91 (2025-03-20): Fix octothorpe comments in ESP32 partitions
The code took each partitions.csv line and cut off everything after a "#"....and then threq out the cut line and parsed the original. This means it would parse commented lines as if they just had a name of "#spiffs", for example.

Actually use the trimmed line in calculations.

Fixes #90
commit 11f20f11 (2025-03-20): Fix octothorpe comments in ESP32 partitions (#91)

The code took each partitions.csv line and cut off everything after
a "#"....and then threq out the cut line and parsed the original.
This means it would parse commented lines as if they just had a
name of "#spiffs", for example.

Actually use the trimmed line in calculations.

Fixes #90