earlephilhower/arduino-littlefs-upload #101: Arduino IDE 2.3.6 & esptool warnings

OK1FWG, 2025-08-27:
Hi,
thanks for your effort creating this useful tool. Great job!
My system throw out these warnings:

```
Warning: Deprecated: Option '--flash_mode' is deprecated. Use '--flash-mode' instead.
Warning: Deprecated: Option '--flash_freq' is deprecated. Use '--flash-freq' instead.
Warning: Deprecated: Option '--flash_size' is deprecated. Use '--flash-size' instead.
Warning: Deprecated: Choice 'default_reset' for option '--before' is deprecated. Use 'default-reset' instead.
Warning: Deprecated: Choice 'hard_reset' for option '--after' is deprecated. Use 'hard-reset' instead.
Warning: Deprecated: Command 'write_flash' is deprecated. Use 'write-flash' instead.
esptool v5.0.0
```

May be, you noticed this already, just FYI.
Thanks & regards,
Willi

zoutepopcorn, 2025-10-07:
Gemini created a python script for mee

```python
#!/usr/bin/env python3
import sys
import os
import subprocess

script_dir = os.path.dirname(os.path.realpath(__file__))
esptool_binary = os.path.join(script_dir, "esptool")
args = sys.argv[1:]
command = [esptool_binary] + args

subprocess.run(command)
```

place it in 
```/home/popcorn/.arduino15/packages/esp32/tools/esptool_py/5.1.0```

```bash
chmod +x esptool.py
```

and I think it will work now?

HamzaYslmn, 2026-04-07:
i solved btw in v2.0.0