earlephilhower/arduino-littlefs-upload #103: Uploading files with long paths causes a buffer overflow.

H3CATE, 2025-08-29:
I am currently building a small project and it seems like having a long file path (about 32 characters I believe) causes a buffer overflow and crashes the upload. Using Arduino IDE 2.3.6 on Fedora 42. 

It crashes during the "Building LittleFS filesystem" with message "*** buffer overflow detected ***: terminated" and results in the uploader trying to upload an empty image.

earlephilhower, 2025-08-29:
Please provide a ZIP of the upload directory that's crashing and what specific board your building on, so that we can reproduce the failure.

H3CATE, 2025-08-29:
Working on an ESP-32 C6 N8 DevKit from Waveshare.

As for the ZIP, I have fixed the issue by shortening filenames, but I recreated a test folder and managed to reproduce the issue:

`LittleFS Filesystem Uploader v1.5.4 -- https://github.com/earlephilhower/arduino-littlefs-upload

 Sketch Path: /home/hecate/Documents/testsketch
   Data Path: /home/hecate/Documents/testsketch/data
      Device: ESP32 series, model esp32c6
Using partition: default
  Partitions: /home/hecate/.arduino15/packages/esp32/hardware/esp32/3.3.0/tools/partitions/default.csv
       Start: 0x290000
         End: 0x3f0000
 Serial Port: /dev/ttyACM1

Building LittleFS filesystem
Command Line: /home/hecate/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933/mklittlefs -c /home/hecate/Documents/testsketch/data -p 256 -b 4096 -s 1441792 /tmp/tmp-16387-vTEeo3CxBrB9-.littlefs.bin
/zfolder1/veryverylongfilenameabovelimits.md
*** buffer overflow detected ***: terminated

Uploading LittleFS filesystem
Command Line: /home/hecate/.arduino15/packages/esp32/tools/esptool_py/5.0.0/esptool --chip esp32c6 --port /dev/ttyACM1 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 /tmp/tmp-16387-vTEeo3CxBrB9-.littlefs.bin
Warning: Deprecated: Option '--flash_mode' is deprecated. Use '--flash-mode' instead.
Warning: Deprecated: Option '--flash_freq' is deprecated. Use '--flash-freq' instead.
Warning: Deprecated: Option '--flash_size' is deprecated. Use '--flash-size' instead.
Warning: Deprecated: Choice 'default_reset' for option '--before' is deprecated. Use 'default-reset' instead.
Warning: Deprecated: Choice 'hard_reset' for option '--after' is deprecated. Use 'hard-reset' instead.
Warning: Deprecated: Command 'write_flash' is deprecated. Use 'write-flash' instead.
esptool v5.0.0
Serial port /dev/ttyACM1:
Connecting...
Connected to ESP32-C6 on /dev/ttyACM1:
Chip type:          ESP32-C6 (QFN40) (revision v0.1)
Features:           Wi-Fi 6, BT 5 (LE), IEEE802.15.4, Single Core + LP Core, 160MHz
Crystal frequency:  40MHz
USB mode:           USB-Serial/JTAG
MAC:                Redacted
BASE MAC:           Redacted
MAC_EXT:            ff:fe

Uploading stub flasher...
Running stub flasher...
Stub flasher running.
Changing baud rate to 921600...
Changed.

Configuring flash size...
Auto-detected flash size: 8MB
Flash will be erased from 0x00290000 to 0x0028ffff...
Warning: '/tmp/tmp-16387-vTEeo3CxBrB9-.littlefs.bin' is empty.

Hard resetting via RTS pin...

Completed upload.`

[data.zip](https://github.com/user-attachments/files/22053362/data.zip)

earlephilhower, 2025-08-29:
Thanks for that, it's very useful.

Your system is running an ancient LittleFS with a 32 character name limit.  About 14 months ago 4.0.0 came out with the limit lifted to LittleFS's max (255 chars IIRC).  The ESP32 and Pico both support the full 255 char limit while the ESP8266 was set to 32 bytes for memory limitation in our Arduino core.  Hence the 32byte limit in old versions.

Are you running a new-ish ESP32 Arduino core?  They should have moved to the 4.0.0/1 in the past year and change.  If not, then we need to open something on their repo and have them pull my newer version.

H3CATE, 2025-08-29:
I have indeed seen that issue during my debugging search on the net... 
I am running the esp32 3.3.0 core installed from the board manager:

[image]

https://github.com/espressif/arduino-esp32

Seems to be the latest version here and on the github page, or am I misunderstanding ?

earlephilhower, 2025-08-29:
No, you're good.  It's Espressif who's busted here, unfortunately.  This plugin just runs the specific core's `mklittlefs` and if the core is wrong, there's nothing to do here.

Looks like Espressif forked my code and built a local copy a long time ago and never updated.  The crash/32b limit is in their own executable, not one from the current GH releases or source code.

At this point I'd recommend opening up an issue on their GH repo or forums asking them to update `mklittlefs` to latest.  Or, at least, to `#define  LFS_NAME_MAX 255` in their old code and rebuild.  Nothing we can do here.

For now, on your own, can manually download and overwrite your installed mklittlefs with the one from https://github.com/earlephilhower/mklittlefs/releases/tag/4.0.2 to use longer filenames.

closed 2025-08-29 by earlephilhower

H3CATE, 2025-08-29:
Ok that's done, thanks a bunch for the help and sorry for bothering you with that ! Have a great day :D